joelsernamoreno/EvilCrowRF-V2 #1: Hi Dear,

HellGateCorp, 2022-01-31:
im really happy to see you are planning a V2 of your great RF Tool. I searched for hours to find a seller who selling the first one and got found one with only 3 items in stock and im in doubt it make sense to buy the old one. So my question is, can you tell me approximately when you planned to sell the V2 and where, also on Tindie? Wanna set my order in the first place to get one, not like the V1 lol. Greetz

joelsernamoreno, 2022-01-31:
Hi!

The official Aliexpress and Tindie store is called April Brother. Do not buy Evil Crow RF from other store, you may be scammed with this

Currently there is no stock of ECRF V1, I think ECRF V1 will not be manufactured again

ECRF V2 will be released after Chinese New Year, the manufacturer is on vacation now. I will let you know when ECRF V2 will be on sale

Thanks!

HellGateCorp, 2022-01-31:
Im really glad to hear that, thanks for your fast answer. 

The seller is a British Biohacking Lab I think they bought it from AprilBother and resell it to the European Market, when you want to know I can tell you the URL and if not you can close that Thread for sure.

Thank you so much for sharing that wonderful Tool to the Community and wish you a wonderful sales start and good balance sheets, I can't wait to place my Order for the half of this week. : )

Greetz

closed 2022-12-16 by joelsernamoreno