joelsernamoreno/EvilCrowRF-V2 #22: for loop exceeds array bounds

buttim, 2023-02-19:
The two for loops in line 1383 and 1391 access the `jammer` array. The cycle index goes from 0 to 10 and is used to index the array at `i` and `i+1` but the array is declared as
`byte jammer[11] = {0xff,0xff,};`
which means the index can only go from 0 to 10.
This can be easily spotted by simply enabling the default level of warnings for the compiler in the IDE settings

joelsernamoreno, 2025-10-17:
Hello

I’ve released a new firmware update.
I’m going to close all existing issues. If you encounter any problems with the new firmware, or if any previous issues haven’t been resolved, please open a new issue.

Thank you!

closed 2025-10-17 by joelsernamoreno

buttim, 2025-10-17:
It looks to me that the code is still wrong. Feel free to reopen the issue if you're so inclined

buttim, 2025-10-18:
Nice! I noticed that the jammer array is no longer 11 bytes long, just 2 now. I guess that was your intention though.