joelsernamoreno/EvilCrowRF-V2 #24: [Proposal] Add support to Flipper Zero .sub files

iordic, 2023-03-05:
Hi Joel,
I was reading your code and I think you can add support to .sub files easily copying the function `sendByte()` (at least for the raw format):
https://github.com/joelsernamoreno/EvilCrowRF-V2/blob/0f64a4cadab4e06791b59aef5354fa9125c761a4/firmware/v1.3.2/EvilCrow-RFv2/EvilCrow-RFv2.ino#L518-L523

Here you have a guide on how to translate the format: [https://pandwarf.com/news/kaiju-supports-flipper-zero-sub-ghz-signal-files/](url)

Reading the header you can obtain the preset config. Then, when the "Raw data" section of file has been correctly processed you can obtain an array of integers as a result. I suppose it should work with:

```c
// expecting an integer array. Ex.: [ -1038, 487, -1044, 511, -1010, 1011]
void sendSub(int *values, int size) {
  for (int i = 0; i < size; i++) {
      digitalWrite(TXPin0, values[i] > 0 ? HIGH : LOW);
      delayMicroseconds(abs(values[i]));
    }
}
```
This would definitely increase the versatility of the evilcrow because there are many files for the zero flipper on the internet. BTW, I'm going to try myself with an ESP32 devboard because i don't have an evilcrowRF device, but i can update you with my results.

nullvoid413, 2023-03-12:
Any updates on this?

iordic, 2023-03-12:
yup! I got it with my ESP32 dev board and my own code taking the part i said above (i don't know why i cannot get cc1101 working properly with evil crow firmware :pensive:).

I had a code to read a file on sd card with path `/subghz/Pure_raw_up.sub` (hardcoded path for now). And i print on Serial this debug info:
```
[SD CARD] Type: SDHC
[SD CARD] Size: 3776 MiB
/subghz/Pure_raw_up.sub
FZ signal -> Freq:433.92, Mod:2, Size:1991.
```
[image]

The code opens the file and parse the lines to retrieve the info to a defined struct. Each line is parsed with this function:
```c
void parseLine(char *line, struct FZSubSignal *fz) {
  const char sep[2] = ":";
  const char values_sep[2] = " ";
  
  char *key = strtok(line, sep);
  char *value;
  
  if (key != NULL) {
      value = strtok(NULL, sep);
      if(!strcmp(key, "Version")) {
          fz->version = atoi(value);
      } else if(!strcmp(key, "Frequency")) {
          fz->frequency = atof(value) / 1000000.0f;
      } else if(!strcmp(key, "Preset")) {
          fz->modulation = 2;   // ASK/OOK; TODO: translate strings like: FuriHalSubGhzPresetOok270Async
      } else if(!strcmp(key, "RAW_Data")) {
          char *pulse = strtok(value, values_sep);
          int i = fz->size;          
          while (pulse != NULL && i < FZ_SUB_MAX_SIZE) {
              fz->data[i] = atoi(pulse);
              pulse = strtok(NULL, values_sep);
              i++;
          }
          fz->size = i;
      }
  }
}
```
The structure for the file data is:
```c
struct FZSubSignal {
    uint8_t version;
    float frequency;
    uint8_t protocol;       // Now: RAW or BinRAW (first format with pulses and second with hex values)
    uint8_t modulation;     // 2-FSK, GFSK, ASK, 4-FSK, MSK
    uint16_t te;
    uint32_t size;
    int32_t data[FZ_SUB_MAX_SIZE];    
};
```
And then do the transmission:
```c
// function definition for sending individual bits with FZ format
void sendBits(int *buff, int length) {
  for (int i = 0; i < length; i++) {
    digitalWrite(GDO0, buff[i] < 0 ? LOW : HIGH);
    delayMicroseconds(abs(buff[i]));
  }
  digitalWrite(GDO0, LOW);
}

// ... here a call to the method with the data parsed ...
pinMode(GDO0,OUTPUT);
ELECHOUSE_cc1101.Init();
ELECHOUSE_cc1101.setModulation(fzSubSignal->modulation);
ELECHOUSE_cc1101.setMHZ(fzSubSignal->frequency);
ELECHOUSE_cc1101.setDeviation(0);
ELECHOUSE_cc1101.SetTx();

sendBits(fzSubSignal->data, fzSubSignal->size); 
```

## Futures
* I'll keep trying to get evilcrow v2 code working on my esp32 board, if i will get it, i'll try to add  a web access to upload or transmit the files.
* It would be great to implement the other formats.
### New BinRAW format
All of detailed here is for the old/common format (Example [here](https://github.com/UberGuidoZ/Flipper/blob/main/Sub-GHz/Vehicles/Tesla/BEST_PORT_OPENER/433.92MHz_AM650_Better_Tesla_Charge_Port_Opener.sub)).
On the other hand, now there is another file format that will have to be implemented (see https://github.com/flipperdevices/flipperzero-firmware/pull/2322).

I've have an example file:
```
Filetype: Flipper SubGhz Key File
Version: 1
Frequency: 433920000
Preset: FuriHalSubGhzPresetOok650Async
Protocol: BinRAW
Bit: 311
TE: 205
Bit_RAW: 287
Data_RAW: 7F FF FF C7 47 44 47 74 44 74 47 74 47 44 74 77 77 44 47 44 44 44 44 74 44 44 44 44 47 44 74 44 47 77 77 77
Bit_RAW: 24
Data_RAW: 00 00 00
```

---
P.S. You can view it working here:

https://user-images.githubusercontent.com/18122773/224566485-1343f570-a2d3-4fa0-b6fb-e6ca236a51db.mp4

1077213d91, 2023-03-20:
> 

Hello, how to use the 'parseLine' function please? My ESP crashes when I try, I must be doing it wrong.

Can you post the complete code please?

iordic, 2023-03-20:
Here you go, a simple code and the testing file. I have a bug with the file, it's opening only at the first time, but the rest of code works properly.

[sparrowRF.ino.txt](https://github.com/joelsernamoreno/EvilCrowRF-V2/files/11021455/sparrowRF.ino.txt)
[purawup.sub.txt](https://github.com/joelsernamoreno/EvilCrowRF-V2/files/11021467/purawup.sub.txt)

P.S: you can avoid using dynamic memory by putting the structure as a global variable.

joelsernamoreno, 2023-03-29:
New firmware to be released soon compatible with flipper zero

joelsernamoreno, 2023-03-29:
@iordic Have you identified your problem with the board? It should work if you are using ESP32 and the correct pins

iordic, 2023-03-29:
Yes, i had it working properly days ago. I could submit a PR if you want.

But now i have problems with RX, reading random values and crashing. Are RxPin0 an RxPin values assigned to GDO2 of module 0 and GDO2 of module 1 respectively?

closed 2023-05-12 by iordic