joelsernamoreno/EvilCrowRF-V2 #29: Cannot change AP name or password on most firmware versions.

trigat, 2023-05-13:
I was having issues with this a month ago.  I decided to come back to it today and I'm still having problems.  If I edit the SSID, password, or wifi_channel in the  .ino, the updates do not reflect on the Evil Crow after uploading the sketch.  I am also not able to see the AP name from any Android phone.  I can however, find the AP on a computer and laptop, allowing me to connect.

After uploading the firmware 1.3.2 sketch, I do have an option to change the AP name from with the browser UI.  After changing the name and applying the change, the EvilCrow reboots but does not actually change the AP name.

I have had luck at times being able to change the AP info after changing the v1.3.1 sketch.  Still no luck finding the AP while on a phone.  I've even used Android apps such as Ubiquiti's WiFiman to search for all APs in my area and it could not see the Evil Crow's AP.

I have tried uploading different firmware sketches from multiple Ubuntu computers.  I've also tried this using the Arduino version 1.8.13, 1.8.19, and 2.1.0.  I have this problem on all machines and Arduino versions.

I've used these same Arduino setups to program ESP32 boards in the past and have not had issues editing AP names and credentials.  Do you happen to know what may be causing this?

AP names I've tried as an example are "Crow" and "EvilCrow."  Password lengths I've tried are 11-15 characters in length.

A few moments ago, I grabbed a fresh laptop that has never had Arduino installed on it.  I went through your installation steps and had the exact same issues.

I'm editing the AP name and password using the following lines in the .ino:

// Config SSID, password and channel
const char* ssid = "BlackCrow";  // Enter your SSID here
const char* password = "passwordhere";  //Enter your Password here
const int wifi_channel = 14; //Enter your preferred Wi-Fi Channel

Any ideas on what may be causing this?  Is the original AP info somehow being cached to memory on the Evil Crow device and not updating with the Arduino sketch?

trigat, 2023-05-13:
Another update.  I used the default WiFi settings to upload firmware v1.1.  This allowed me to finally change the AP name to "Evil Crow RF v2."  The name has been stuck on "crow" since last month.  I then immediately uploaded the v1.3.2 sketch with the default .ino.  I copied the SD files over to the SD card after each upload.

I then went to http://192.168.4.1 in the browser and selected the ECRF Config > WiFi Config menu.

I wrote an SSID name of "BlackCrow", updated the passwords to 14 characters in length, and hit Apply.

This worked.  The AP name is now up-to-date.  I guess the AP info should not be updated from within the .ino file.

I'm still not able to view the AP using a phone.  I can view the Crow's AP from Windows and Ubuntu laptops.  Do you happen to know what may be causing this?  I have another ESP8266 and ESP32 board that I can currently connect to using a phone.

joelsernamoreno, 2023-05-18:
Hello

This is due to the WiFi channel used. You can change this in the firmware and reflash the firmware on the device

This is default, change this
const int wifi_channel = 12

trigat, 2023-05-21:
Thank you.  I should have tested that.

closed 2023-05-21 by trigat