joelsernamoreno/EvilCrowRF-V2 #33: Evil crow web page not loading on 192.168.4.1

NicheSkill, 2023-10-06:
I am unable to load Web page of Evil crow. I can ping to the ip.
I can see Evil Crow SSID, I can connect to it.
I have copied files to SD card. 
SD card is 32Gb and formatted using FAT32
I tried changeing SD card module and also confirmed SD card is  working fine.
Below are connections to SD Card.
ESP32          SD Card.
==============
Pin 22         CS
Pin 23         MOSI
Pin 18         SCK
Pin 19         MISO

My esp is "DOIT ESP32 DEVKIT V1"
I am not getting any error during compilation.
What could be the reason. ???

joelsernamoreno, 2023-10-06:
Hello!

How did you copy the files to the microSD card. You have to copy the complete folders

In the root of the microSD card must be the HTML and URH folders

If this does not work, try another microSD card, some microSD cards do not work. You can try a smaller microSD card

wpa-2, 2023-10-07:
If your connecting via a mobile device make sure data connection is disabled before tyring to connect. 
I had that issue./

joelsernamoreno, 2023-10-07:
@wpa-2 Mm... on my phones is not necessary: Samsung Note 10+ and Samsung Galaxy S8

Snifi0, 2023-10-08:
> @wpa-2 Mm... on my phones is not necessary: Samsung Note 10+ and Samsung Galaxy S8

i have the same issue, i made a custom pcb and my cs pin is IO13, thats not a problem, right?

NicheSkill, 2023-10-08:
Solved! I was supplying  3.3v to SD card module. The moment I changed it to 5 volts it fixed immediately!!!!

closed 2023-10-11 by NicheSkill