joelsernamoreno/EvilCrowRF-V2 #35: Cannot upload

in4matix, 2023-12-11:
Stuck at step 15. 

```
/home/daniel/Arduino/EvilCrowRF-V2/firmware/v1.3.2/EvilCrow-RFv2/EvilCrow-RFv2.ino:4:22: fatal error: AsyncTCP.h: No such file or directory
compilation terminated.

exit status 1

Compilation error: AsyncTCP.h: No such file or directory
```

joelsernamoreno, 2023-12-20:
Hello!

The error indicates that you do not have the Async library downloaded. Read the documentation of the repository and follow the steps indicated in the repository

joelsernamoreno, 2023-12-20:
You have to download and unzip the library in the directory /home/daniel/Arduino/libraries/

in4matix, 2023-12-20:
Hi,
Thank you for the reply. Pardon my ignorance, I am fairly new to Linux. Must all of them be cloned into /home/daniel/Arduino/libraries/ ? Or just Async?

in4matix, 2023-12-20:
Currently I have the all in  /home/daniel/Arduino/

joelsernamoreno, 2023-12-20:
The libraries directory is /home/daniel/Arduino/libraries/

Here you have to put all the libraries used by Evil Crow RF V2. The firmware (.ino) can be in another location

in4matix, 2023-12-20:
Thank you very much !

closed 2024-01-17 by joelsernamoreno