joelsernamoreno/EvilCrowRF-V2 #36: "Unexpected error: ESP is not in flash boot mode. If your board has a flashing pin, try again while keeping it pressed."

ronaldosegundojr, 2023-12-19:
I purchased one of these through aliexpress, when I arrived I prepared the SD card with the files and when I plugged it in I saw that it was already configured with the EvilCrow_v2 wifi network
However, when accessing 192.168.4.1, the page loads infinitely.
So I thought about updating the Firmware, but when updating the Firmware I receive the following errors:

Error 1: 
"Unexpected error: ESP is not in flash boot mode. If your board has a flashing pin, try again while keeping it pressed."  

When I press the EN button:
Error 2:
Using 'COM3' as serial port.
Connecting......................................
Unexpected error: ESP Chip Auto-Detection failed: Failed to connect to Espressif device: No serial data received.
For troubleshooting steps visit: https://github.com/espressif/esptool#troubleshooting

When I press the EN button several times
Error 3:
Using 'COM3' as serial port.
Unexpected error: could not open port 'COM3': PermissionError(13, 'Acesso negado.', None, 5)

esphome-flasher on Windows 11, as administrator.

I also tested an esp boot via browser, but it didn't work either.

I've done several tests, I've put it on different USBs, 2.0, 3.0, with cable, without cable, I've tested with microSD, without microSD and several continuously without being able to update the firmware.
Can someone help me?

joelsernamoreno, 2023-12-20:
Hello!

Check that the port is recognized correctly, maybe you have to install the USB-UART drivers if you are using Windows to flash the firmware

ronaldosegundojr, 2023-12-23:
Hello Joel!! Are you fine?

Thanks for the answer!

It was actually a driver issue that was preventing the Arduino IDE from writing the firmware.

With your tip I managed to solve this problem, but now I'm having another problem:

I managed to recover the firmware, but when accessing 192.168.4.1 there is just a black page, without an error message.

But if I use 192.168.4.1/update I can access an ElegantOTA page normally, but the index.html home page and other .html pages do not load.

I've tried reinstalling the firmware several times, I've tested other SD cards and I still have the same problem...

Do you know how to help me?

The idea is to use this Evil Crow as a jammer,

CernaKocka, 2024-01-08:
> Hello!
> 
> Check that the port is recognized correctly, maybe you have to install the USB-UART drivers if you are using Windows to flash the firmware

Hey, I have the same issue but downloading usb-uart drivers didn't work (device manager says it's incompatible and won't let me install)

joelsernamoreno, 2024-01-17:
> Hola joel!! ¿Estás bien?
> 
> ¡Gracias por la respuesta!
> 
> En realidad, era un problema de controlador que impedía que el IDE de Arduino escribiera el firmware.
> 
> Con tu consejo logré resolver este problema, pero ahora tengo otro problema:
> 
> Logré recuperar el firmware, pero al acceder a 192.168.4.1 solo aparece una página en negro, sin mensaje de error.
> 
> Pero si uso 192.168.4.1/update puedo acceder a una página de ElegantOTA normalmente, pero la página de inicio index.html y otras páginas .html no se cargan.
> 
> Intenté reinstalar el firmware varias veces, probé otras tarjetas SD y sigo teniendo el mismo problema...
> 
> Intenté hacer ping a 192.168.4.1 y responde al ping
> 
> ¿Sabes cómo ayudarme?
> 
> La idea es usar este Evil Crow como jammer, tengo otro Evil Crow que está configurado correctamente, pero este tiene este problema de no cargar el contenido de lo que hay en la tarjeta sd, incluso he probado la tarjeta sd de Evil Crow. que está funcionando, pero no funcionó...
> 
> Necesito ayuda para resolver esto

I understand that this is the same problem as indicated in the other issue. You have a faulty crow on the SD

joelsernamoreno, 2024-01-17:
> > ¡Hola!
> > Comprueba que el puerto se reconoce correctamente, tal vez tengas que instalar los controladores USB-UART si estás usando Windows para actualizar el firmware.
> 
> Ey@joelsernamoreno, Tengo el mismo problema pero la descarga de los controladores USB-UART no funcionó (el administrador de dispositivos dice que es incompatible y no me deja instalarlo).

Hello

If you have not solved the problem let me know. Give me a few days and I'll look for a driver link, I'm quite busy now. @ronaldosegundojr can indicate the link of the drivers he has used

CernaKocka, 2024-01-18:
> Hello
>
> If you have not solved the problem let me know. Give me a few days and I'll look for a driver link, I'm quite busy now. @ronaldosegundojr can indicate the link of the drivers he has used

I haven't solved it yet

hackerethicall, 2024-12-20:
Hey @joelsernamoreno @CernaKocka I have the same issue but downloading usb-uart drivers didn't work

joelsernamoreno, 2025-10-17:
Hello

I’ve released a new firmware update.
I’m going to close all existing issues. If you encounter any problems with the new firmware, or if any previous issues haven’t been resolved, please open a new issue.

Thank you!

closed 2025-10-17 by joelsernamoreno