joelsernamoreno/EvilCrowRF-V2 #37: SD Card Not Recognized

ronaldosegundojr, 2023-12-24:
Hello everything is fine?

I recently acquired 2 Evil Crow to study, but I'm racking my brain to solve a problem.

One of my evil crow is working correctly.

The other one is not recognizing the SD card. I even changed the SD cards on both, with the same files, but it still didn't work.

I've already updated the firmware, went back to the old one, updated it again, tested it on other computers and nothing...
Everything indicates that the SD card is not being recognized, what can I do?

I've already tested it with 4 different sds cards, 2gb, 2gb, 8gb and 32gb

Both with the HTML and URH folders.

thebenjibenji, 2023-12-24:
I have this issue, in the logs it said mounting failed, I don’t have them at the moment but I can supply them later.

joelsernamoreno, 2023-12-24:
Hello!

If one device is working properly and the other device is not working with the same card, it may be a faulty device. Try some basic arduino example to read the microSD card. If this doesn't work, talk to the distributor to send you another device

ronaldosegundojr, 2023-12-26:
Thanks Joel, I thought it could be something else, but what you said made perfect sense, it must be a defect.

Many thanks for the reply!

Have a great day!

ronaldosegundojr, 2023-12-26:
Unfortunately, I have already passed the product exchange deadline, so I thought about injecting the contents of the HTML folder into the firmware itself, at least to use the Jammer function, since the other evil crow is working

joelsernamoreno, 2024-01-17:
> Desafortunadamente, ya pasé la fecha límite para el cambio de producto, así que pensé en inyectar el contenido de la carpeta HTML en el firmware, al menos para usar la función Jammer, ya que el otro cuervo malvado está trabajando con todas las funciones. Si eso funciona, puedo enviar una solicitud de extracción, podría ayudar a alguien que tenga el mismo problema.
> 
> ¡gracias!

Feel free to send PR

Check the Evil Crow RF V1 firmware in the other repository. Evil Crow RF V1 does not have microSD and loads everything from memory, you can use this as a base for V2

Note that the CC1101 pins on Evil Crow RF V1 are different, remember to use the correct V2 pins :)

closed 2024-01-17 by joelsernamoreno