joelsernamoreno/EvilCrowRF-V2 #38: Booting in Downloading Mode

rmivdc, 2024-01-03:
Hi, 
i'm actually stuck in a bootloop :

...

> rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
> flash read err, 1000
> main.c 342
> ets Jan  2 2023 00:22:57
> 
> rst:0x10 (RTCWDT_RTC_RESET),boot:0x1b (SPI_FAST_FLASH_BOOT)
> flash read err, 1000
> main.c 342
> ets Jan  2 2023 00:22:57

...

I've tried multiple flashing solutions from ESP but still can't get that downloading mode :

_ESP Chip Auto-Detection failed: Failed to connect to Espressif device: Invalid head of packet (0x65)_

Which pins do i have to short on the board to enter in this state ?

Thank you

joelsernamoreno, 2024-01-17:
> Hola, en realidad estoy atrapado en un bootloop:
> 
> ...
> 
> > primero: 0x10 (RTCWDT_RTC_RESET), arranque: 0x13 (SPI_FAST_FLASH_BOOT) 
> > error de lectura flash, 1000 
> > main.c 342 
> > ets 2 de enero de 2023 00:22:57
> > primero: 0x10 (RTCWDT_RTC_RESET), arranque: 0x1b (SPI_FAST_FLASH_BOOT) 
> > error de lectura flash, 1000 
> > main.c 342 
> > ets 2 de enero de 2023 00:22:57
> 
> ...
> 
> Probé varias soluciones de actualización de ESP pero todavía no puedo obtener ese modo de descarga:
> 
> _Error de detección automática del chip ESP: no se pudo conectar al dispositivo Espressif: encabezado de paquete no válido (0x65)_
> 
> ¿Qué pines tengo que acortar en el tablero para entrar en este estado?
> 
> Gracias

Hello!

What operating system are you using? In Ubuntu it works fine, you don't have to press any buttons or do any other things.

In debian it gave some problems many times and it doesn't enter in that mode.

I think in windows it should also work correctly

rmivdc, 2024-01-23:
Ok thank you Joel, i'm using Fedora 39, it seems it was a driver issue (?)...
i had a success by using a wire and shorting the gpio9 of the esp32-c3 to GND while resetting the chip :
https://docs.espressif.com/projects/esptool/en/latest/esp32c3/advanced-topics/boot-mode-selection.html

closed 2024-01-23 by rmivdc

Vershik79, 2025-12-31:
Sorry, I ran into the same problem, can you tell me which contact you closed to the ground, number?