joelsernamoreno/EvilCrowRF-V2 #41: Discord (again)

MathewsCarioca, 2024-03-10:
Hey, I'm new to hardware hacking in general and I'm trying to learn more about it. I've bought the Evil Crow RF v2, and I want to buy an RTL-SDR! Unfortunately, the Discord link is not working :( and I was hoping to use it as a community to learn more about the possibilities of using the Evil Crow.

joelsernamoreno, 2024-03-10:
Hi @MathewsCarioca 

https://discord.com/invite/cGP82zg8

Hard2Borrow, 2024-09-02:
Hello, Link above to the Discord along with the 1 in the Readme has expired- pls update.

wpa-2, 2024-09-03:
> Hello, Link above to the Discord along with the 1 in the Readme has expired- pls update.

try 
https://discord.gg/DZnZEZtQUf

joelsernamoreno, 2025-10-17:
Hello

I’ve released a new firmware update.
I’m going to close all existing issues. If you encounter any problems with the new firmware, or if any previous issues haven’t been resolved, please open a new issue.

Thank you!