joelsernamoreno/EvilCrowRF-V2 #42: WARNING: Failed to communicate with the flash chip, read/write operations will fail. Try checking the chip connections or removing any other hardware connected to IOs.

twertybog, 2024-03-18:
Hello!
I purchased an evil crow rf v2 on aliexpress after arriving I tried to flash it with firmware but error appeared. In arduino ide error: 
```
Sketch uses 940046 bytes (71%) of program storage space. Maximum is 1310720 bytes.
Global variables use 90720 bytes (27%) of dynamic memory, leaving 236960 bytes for local variables. Maximum is 327680 bytes.
esptool.py v3.0-dev
Serial port /dev/ttyUSB0
Connecting........_____....._____....._____....._
Chip is ESP32-PICO-D4 (revision 1)
Features: WiFi, BT, Dual Core, 240MHz, Embedded Flash, VRef calibration in efuse, Coding Scheme None
Crystal is 40MHz
MAC: d4:d4:da:97:8d:f8
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 921600
Changed.
Configuring flash size...
Warning: Could not auto-detect Flash size (FlashID=0x0, SizeID=0x0), defaulting to 4MB
Compressed 8192 bytes to 47...
Writing at 0x0000e000... (100 %)
Wrote 8192 bytes (47 compressed) at 0x0000e000 in 0.0 seconds (effective 19910.0 kbit/s)...
```
After that, I tried to erase flash with esptool
```
esptool.py --chip esp32 --port /dev/ttyUSB0 erase_flash
```
```
esptool.py v4.7.0
Serial port /dev/ttyUSB0
Connecting....
Chip is ESP32-PICO-D4 (revision v1.0)
Features: WiFi, BT, Dual Core, 240MHz, Embedded Flash, VRef calibration in efuse, Coding Scheme None
Crystal is 40MHz
MAC: d4:d4:da:97:8d:f8
Uploading stub...
Running stub...
Stub running...
WARNING: Failed to communicate with the flash chip, read/write operations will fail. Try checking the chip connections or removing any other hardware connected to IOs.
Erasing flash (this may take a while)...

A fatal error occurred: Packet content transfer stopped (received 3016 bytes)
```
And got info about flash
```
esptool.py --chip esp32 --port /dev/ttyUSB0 flash_id
```
```
esptool.py v4.7.0
Serial port /dev/ttyUSB0
Connecting....
Chip is ESP32-PICO-D4 (revision v1.0)
Features: WiFi, BT, Dual Core, 240MHz, Embedded Flash, VRef calibration in efuse, Coding Scheme None
Crystal is 40MHz
MAC: d4:d4:da:97:8d:f8
Uploading stub...
Running stub...
Stub running...
WARNING: Failed to communicate with the flash chip, read/write operations will fail. Try checking the chip connections or removing any other hardware connected to IOs.
Manufacturer: 00
Device: 0000
Detected flash size: Unknown
Hard resetting via RTS pin...
```

joelsernamoreno, 2024-03-18:
Hello @twertybog 

What is your OS? In debian it gives some problems, in Ubuntu it should work correctly

twertybog, 2024-03-18:
> Hello @twertybog
> 
> What is your OS? In debian it gives some problems, in Ubuntu it should work correctly

Thanks for response.
I use Arch. I will try Ubuntu for flashing.

lrosario, 2024-12-02:
@joelsernamoreno can u help me?

joelsernamoreno, 2024-12-02:
Hi @lrosario 

I don't have macOS, I can't test this. Can you try it on Windows or Ubuntu?

lrosario, 2025-01-03:
i tried on Ubuntu, same error.

samxplogs, 2025-03-27:
@lrosario I am running in similar issue, I even tried esp home and here is the issue I've got: Installation failed: Error: Packet content transfer stopped

LED1 stay in RED since the erasing process

I am wondering if this is not bricked. I do see the network ECRF, I can join and ping the gateway but nothing else.

I even got myself a second one but it seems DOA despite the fact I can flash it, no SSID found