joelsernamoreno/EvilCrowRF-V2 #43: Discord

matteasd, 2024-03-20:
Hello!

Is it possible to get the discord link? :)

joelsernamoreno, 2024-03-20:
Hi @matteasd 

https://discord.com/invite/7AtRFDga

crazyfox-ua, 2024-04-10:
@joelsernamoreno one more, please

crystalclear30, 2024-04-18:
May I get an invite to the Discord group please?  Just received my device and having trouble uploading sketches.  I've apparently bricked my device by changing to "Station" mode as well :(

frederikkunze, 2024-11-25:
Discord links in the description are not valid. Is the Discord not active anymore or do they need to get updated?

joelsernamoreno, 2024-11-26:
Hello @frederikkunze 

Discord: https://discord.gg/CV5hD4wx

overlordtm, 2024-12-29:
Can i get discord invite please? Current one expired.

It should be possible to create a permanent invite: https://support.discord.com/hc/en-us/articles/208866998-Invites-101

wpa-2, 2024-12-29:
> Can i get discord invite please? Current one expired.
> 
> It should be possible to create a permanent invite: https://support.discord.com/hc/en-us/articles/208866998-Invites-101

https://discord.gg/G6ZavyZf
yes but permanent ones do cause issues

joelsernamoreno, 2024-12-29:
Hi @overlordtm 

https://discord.gg/r3Sy7jm3

cinouw, 2025-04-04:
Hi i would like to join the discord server can i get a invitation link

joelsernamoreno, 2025-04-04:
Hi @cinouw 

https://discord.gg/u8R7VN4d

cinouw, 2025-04-04:
> Hi @cinouw 
> 
> https://discord.gg/u8R7VN4d

It's saying that it is expired 🥲

joelsernamoreno, 2025-04-04:
> > Hi [@cinouw](https://github.com/cinouw)
> > https://discord.gg/u8R7VN4d
> 
> It's saying that it is expired 🥲

https://discord.gg/Nu6Np8YXDH

frederikkunze, 2025-04-04:
@joelsernamoreno you could also generate a link without an expiration date.
https://support.discord.com/hc/en-us/articles/208866998-Invites-101#h_01H2DMMG355MV7GJDQ9F65SDVG

joelsernamoreno, 2025-04-04:
Hi @frederikkunze 

The last link is configured without expiration

Thanks

joelsernamoreno, 2025-10-17:
Hello

I’ve released a new firmware update.
I’m going to close all existing issues. If you encounter any problems with the new firmware, or if any previous issues haven’t been resolved, please open a new issue.

Thank you!

closed 2025-10-17 by joelsernamoreno