joelsernamoreno/EvilCrowRF-V2 #44: Problem reading SD Card

jeff-azevedo, 2024-04-11:
Hi all! I bought a Evil Crow to start learning hardware hacking, but got stuck in the last step of the tutorial
The board's webserver simply doesn't go up. Searching for issues I find out that the SD card was not detected.

I tried the following code:

#include <SD.h>
void setup() {
Serial.begin(38400);
for (int csPin = 5; csPin <= 32; csPin++) {
if (SD.begin(csPin)) {
Serial.print("SD card detected on CS pin: ");
Serial.println(csPin);
} else {
Serial.print("No SD card found on CS pin: ");
Serial.println(csPin);
}
}
}
void loop() {}

But never got success reading the SD card.

I tried:

- Formating the SD card in FAT32 and FAT16
- Changing SD card
- Creating a file in the SD card using my PC and tried read it in the uploaded firmware

If I dont solve this issue till tomorrow I'm considering to return the board and ask for refund.

Is it a board's problem or software? Any thoughts or tips on how to solve this?

volca, 2024-04-15:
Please check [the source code v1.3.2](https://github.com/joelsernamoreno/EvilCrowRF-V2/blob/main/firmware/v1.3.2/EvilCrow-RFv2/EvilCrow-RFv2.ino#L20)

```
#define SD_SCLK 18
#define SD_MISO 19
#define SD_MOSI 23
#define SD_SS   22

SPIClass sdspi(VSPI);

// .....

void setup() {
  // ...
  sdspi.begin(18, 19, 23, 22);
  SD.begin(22, sdspi);
}
```

Did you try the default firmware? There's a compiled firmware file in the same directory.

jeff-azevedo, 2024-04-15:
Hi, thank you for answering.

I tried the default firmware (1.3.2), even the older versions (1.3.1 and 1.3.0)
The webserver actually goes up but all it returns is 404 cause it doesnt find the pages to serve (located in the SD Card)

The below code is adapted to only check if SD card is being read:
```
#include <SD.h>

#define SD_SCLK 18
#define SD_MISO 19
#define SD_MOSI 23
#define SD_SS   22

SPIClass sdspi(VSPI);

void setup() {
Serial.begin(38400);

  sdspi.begin(18, 19, 23, 22);

  if (SD.begin(22, sdspi)) {
      Serial.print("SD card detected on CS pin");
  } else {
      Serial.print("No SD card found on CS pin");
  }
}

void loop() {}
```
Always returns "No SD card found on CS pin"
Manufacturer says the board works but none of the trials worked.

volca, 2024-04-15:
Did you do the two steps? 

* Copy the `EvilCrowRF-V2/firmware/v1.3.2/SD/HTML` folder to a MicroSD card.
* Copy the `EvilCrowRF-V2/firmware/v1.3.2/SD/URH` folder to a MicroSD card.

Did you find the HTML and URH folder at the root folder?

jeff-azevedo, 2024-04-15:
I did.

[image]

NeedMoreCoffee12, 2024-05-17:
Hi , did you manage to solve it?

jeff-azevedo, 2024-05-30:
I sent back to the manufacturer and was refunded.

closed 2024-05-30 by jeff-azevedo

NeedMoreCoffee12, 2024-05-30:
Unluckily is too late for the aliexpress manufacturer :(