joelsernamoreno/EvilCrowRF-V2 #45: Bricked in Station Mode - is there any way to recover?

crystalclear30, 2024-04-18:
I've accidentally changed my EvilCrowRF-V2 from AP to Station mode without entering any connection settings (blank).  Apparently, this bricks the device.  I cannot connect to the device any longer via wifi.  Is there any way to recover or reset the device with factory firmware settings?  Thank you.

5ubterranean, 2024-07-04:
before uploading the firmware to the device go to the "readConfigWifi" function and modify the line that says "storage_status =1;" to "storage_status =0;", this will make the device ignore any Wifi configuration you have made, then you can set up the Wifi configuration correctly and upload the firmware again but without changes.

lukasz-kwasniewski, 2024-11-22:
what if I deleted whole memory so bootloader and partition too?