joelsernamoreno/EvilCrowRF-V2 #46: schematic / board layout?

mikebanz, 2024-06-12:
Hi Joel

I got a new battery for my ecrfv2 and stupidly assumed the colours on the lead were.

Now the magiscsmoke has come out of a couple of what I believe to be SOT transistors. Is there a circuit diagram and board layout available so I can fix the problem. I googled around and found a schematic for the original board (at least part of it), but I'm no wiser.

Are these resources publicly available somewhere ?

Cheers

Mmike

mikebanz, 2024-12-01:
>    Hello @mikebanz
>    Sorry for taking so long to reply, there are a lot of people asking for support and I don't have so much free time to reply quickly
>    I will check what are the components and I will tell you this in a few days
>    Thanks

Hi Joel,
   
  
    
   
  
   No worries, thanks for taking a look at it.
   
  
    
   
  
   Cheers
   
  
    
   
  
   Mike

joelsernamoreno, 2024-12-02:
Hi @mikebanz 

The component is MOSFET AO3401,ESD CESD5V0AP

joelsernamoreno, 2025-10-17:
Hello

I’ve released a new firmware update.
I’m going to close all existing issues. If you encounter any problems with the new firmware, or if any previous issues haven’t been resolved, please open a new issue.

Thank you!