joelsernamoreno/EvilCrowRF-V2 #47: error compiling

kavemang, 2024-07-08:
Hi getting the following error when trying to compile in IDE
```

Arduino: 1.8.19 (Windows Store 1.8.57.0) (Windows 10), Board: "ESP32 Dev Module, Disabled, Disabled, Default 4MB with spiffs (1.2MB APP/1.5MB SPIFFS), 80MHz (WiFi/BT), DIO, 40MHz, 4MB (32Mb), 921600, Core 1, Core 1, None, Disabled, Disabled"

Multiple libraries were found for "WiFiClient.h"

 Used: C:\Users\user\OneDrive\Documents\ArduinoData\packages\esp32\hardware\esp32\3.0.2\libraries\WiFi

In file included from C:\Users\user\Downloads\EvilCrowRF-V2-1.3.2\EvilCrowRF-V2-1.3.2\firmware\v1.3.2\EvilCrow-RFv2\EvilCrow-RFv2.ino:6:

 Not used: C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.57.0_x86__mdqgnx93n4wtt\libraries\WiFi

C:\Users\user\OneDrive\Documents\Arduino\libraries\AsyncElegantOTA\src/AsyncElegantOTA.h:4:2: error: #error AsyncElegantOTA library is deprecated, Please consider migrating to newer ElegantOTA library which now comes with an async mode. Learn More: https:

    4 | #error AsyncElegantOTA library is deprecated, Please consider migrating to newer ElegantOTA library which now comes with an async mode. Learn More: https://docs.elegantota.pro/async-mode/

      |  ^~~~~

C:\Users\user\OneDrive\Documents\Arduino\libraries\AsyncElegantOTA\src/AsyncElegantOTA.h:18:14: fatal error: esp_int_wdt.h: No such file or directory

   18 |     #include "esp_int_wdt.h"

      |              ^~~~~~~~~~~~~~~

compilation terminated.

exit status 1

Error compiling for board ESP32 Dev Module.

```

Oliver0804, 2024-08-12:
[image]
[image]
In the fourth line of AsyncElegantOTA.cpp:
```
#error AsyncElegantOTA library is deprecated, Please consider migrating to newer ElegantOTA library which now comes with an async mode. Learn More: https://docs.elegantota.pro/async-mode/

```
Simply remove it.

joelsernamoreno, 2024-08-12:
Hello

I've been busy these months

Looks like the library has been changed to a new version:

https://github.com/ayushsharma82/ElegantOTA

Another option is to download the new library and add it to the firmware

joelsernamoreno, 2024-10-18:
Try the new firmware version 

The new firmware is compatible with the latest version of arduino ide and with the latest version of espressif esp32.

Remove all old libraries, remove the ECRF custom wifi settings and follow all the steps in the "Installation" section.

closed 2024-11-22 by joelsernamoreno