joelsernamoreno/EvilCrowRF-V2 #50: wdt won't allow boot after 1.3.3

lukeswitz, 2024-11-06:
Had no issues with the watchdog until this build:
```
09:15:42.871 -> E (55874) task_wdt: esp_task_wdt_reset(763): task not found
09:15:42.903 -> E (55883) task_wdt: esp_task_wdt_reset(763): task not found
09:15:42.903 -> E (55921) task_wdt: esp_task_wdt_reset(763): task not found
09:15:42.935 -> E (55922) task_wdt: esp_task_wdt_reset(763): task not found
```

lukeswitz, 2024-11-06:
seems to shut up after a bit... odd.

closed 2024-11-06 by lukeswitz

joelsernamoreno, 2024-11-06:
Yes, after connecting and disconnecting the device a couple of times... it doesn't seem to give error

Anyway, if it happens again try this:

Edit AsyncTCP/src/AsyncTCP.h:

#define CONFIG_ASYNC_TCP_USE_WDT 1

I have tried this and the firmware doesn't break, but I don't know if it solves wtd. In previous versions WDT 1 broke the firmware