joelsernamoreno/EvilCrowRF-V2 #51: Compilation Error

korn-M, 2024-11-10:
Keep receiving the following error when trying to compile 1.3.3 EvilCrowRF-V2 using Ardiuno 2.3.3. Also using the additional board : https://espressif.github.io/arduino-esp32/package_esp32_index.json

D:\EvilCrowRF-V2-1.3.3\firmware\v1.3.3\EvilCrow-RFv2\EvilCrow-RFv2.ino:10:10: fatal error: SPIFFSEditor.h: No such file or directory
   10 | #include <SPIFFSEditor.h>
      |          ^~~~~~~~~~~~~~~~
compilation terminated.
exit status 1
Compilation error: SPIFFSEditor.h: No such file or directory

I've followed all your steps, but I don't find the SPIFFSEditor.h in the path. Not sure why it is looking for it in EvilCrow-RFv2.ino, it is not like the file can be uploaded there. Right?  Just a newbe obviously.  Thanks in advance for any assistance.

Leftfi3ld, 2024-11-10:
Im running into the same error trying to upload the sketchbook. Im using the 1.3.3 firmware with espressiff 3.0.5 and then all the libraries in the readme with the .h files updated in their .zips. here is the full error upon upload to the board.

/home/sdv/Desktop/EVILCROWWW/EvilCrowRF-V2-1.3.3/firmware/v1.3.3/EvilCrow-RFv2/EvilCrow-RFv2.ino:14:10: fatal error: SPIFFSEditor.h: No such file or directory
   14 | #include <SPIFFSEditor.h>
      |          ^~~~~~~~~~~~~~~~
compilation terminated.
exit status 1

Compilation error: SPIFFSEditor.h: No such file or directory

Korthon, 2024-11-11:
Thank you for the feedback. I checked it’s not in two places. I’m still having the issue.

Korthon, 2024-11-11:
Has the developer abandoned this program? I'm sure you're not the only one that's read this. It's very strange that all files are downloaded. SP web server has spiff editor, but Arduino can't find file or directory. Very strange.

Leftfi3ld, 2024-11-12:
No the developer is still here he just updated the firmware like 2 weeks ago to fix it for us all.

Post your error. I found my lack in arduino/c+ experience was responsible for my problems. Compiling and uploading.

Korthon, 2024-11-13:
I have no C+ experience so this is the error that I get and I don’t understand why it can’t find the spiffs editor because it’s in the library. I feel that I’ve done everything in the list correctly. But obviously not. I guess maybe I need a more step-by-step detail for my lack of experience. Thanks again for your feedback. Totally appreciated.

home/sdv/Desktop/EVILCROWWW/EvilCrowRF-V2-1.3.3/firmware/v1.3.3/EvilCrow-RFv2/EvilCrow-RFv2.ino:14:10: fatal error: SPIFFSEditor.h: No such file or directory
14 | #include <SPIFFSEditor.h>
| ^~~~~~~~~~~~~~~~
compilation terminated.
exit status 1

Compilation error: SPIFFSEditor.h: No such file or directory

compilation terminated.
exit status 1

joelsernamoreno, 2024-11-21:
Hello

Your error indicates that the SPIFFSEditor.h file does not exist

SPIFFSEditor.h is a component of the ESPAsyncWebServer library. You have to download the ESPAsyncWebServer library in your Arduino library directory as indicated in the documentation of the repository

Korthon, 2024-11-23:
Thanks for the info, ESPAasyncwebserver 3.3.22 is installed.

SoLongSidekick, 2024-11-24:
As to why you and I were missing the library file I don't know, but if you google the filename and download the SPIFFSEditor.h file you can manually move it into your EvilCrowv2 folder inside your Arduino folder. 

I mean, I'm also hitting a compilation error (in case you run into it and I've solved it by then, it's "Compilation error: 'virtual bool SPIFFSEditor::canHandle(AsyncWebServerRequest*)' marked 'override', but does not override") but I got past the one you're running into by doing the above.

Korthon, 2024-11-24:
OK, I'll give that a try. Thanks for the info. Appreciate that.

Korthon, 2024-11-25:
I’ve completed numerous projects without encountering any issues. However, this particular project presents a significant challenge for me.

joelsernamoreno, 2024-11-25:
Hello @Korthon 

Can you attach an image of your arduino library directory and an image of the ESPAsyncWebServer library contents?

Korthon, 2024-11-25:
[image]

Korthon, 2024-11-25:
[image]

Korthon, 2024-11-25:
[image]

Korthon, 2024-11-25:
Sorry I couldn't get a snapshot with my laptop

Leftfi3ld, 2024-11-29:
I would recommend adding as many of the libraries possible with the library manager. [image]

Korthon, 2024-12-01:
I downloaded as many as I could. no luck. Thank you for the suggestion.

Leftfi3ld, 2024-12-02:
I was in the same boat as you and I was fried over it. You need to delete
all the libraries, delete your sketchbook. Delete the zip files and then
start from scratch...also look up the H-Rat firmware. It's pretty.

tcccorp, 2025-01-26:
hello, 

i had exactly the same issue. 

after several tries, I removed  everything about esp32, installed board esp32 by espressif  V3.0.5 and installed every lib with git clone ( even on windows) . 

after files modifications ( AsyncTCP.h and ElegantOTA.h) and use right parameters, installation was done with success :)

Korthon, 2025-01-26:
Thanks, i'll give it a go.

Korthon, 2025-04-28:
Finally had time to work on evil crow. Started from scratch formatted SD card went to GitHub. downloaded https://github.com/h-RAT/EvilCrowRF_Custom_Firmware_CC1101_FlipperZero
onto Card. Then I downloaded esphomeflasher. https://github.com/h-RAT/EvilCrowRF_Custom_Firmware_CC1101_FlipperZero

When you go to download onto the SD card, extract files then just download three files from the SD card file. When you flash the evil crow with Home flasher download the complete extract file.