joelsernamoreno/EvilCrowRF-V2 #55: no gateway reachable

tcccorp, 2025-01-03:
hello,  

i upgraded my Evilcrow V2 with the firmware 1.3.3 and there is an issue

I'm able to see the ssid of the Evilcrow, I'm able to connect,  there is an IPv4 ( 192.168.4.2)  with a gateway 192.168.4.1.

I'm not able to ping 192.168.4.1 from my host and there is no arp address, so I'm not able to use my browser. 

do you know I can debug this trouble ?

Thanks

senizokaye, 2025-01-22:
I've had the same issue. 
Solved for me, when I downgraded the board version to 3.0.5
Here is my board and library versions, which works for me normally:
[image]
[image]

tcccorp, 2025-01-26:
hello, 

thanks a lot for your answer and to be honest, shame on me because use 3.0.5 version for esp32 is on the readme :(

tcccorp

closed 2025-01-26 by tcccorp