joelsernamoreno/EvilCrowRF-V2 #56: please how fixe this issue SPIFFSEditor.h

salahgeek, 2025-02-18:
C:\Users\Deamon\Desktop\EvilCrowRFV2\zip\EvilCrowRF-V2-main\EvilCrowRF-V2-main\firmware\v1.3.3\EvilCrow-RFv2\EvilCrow-RFv2.ino:10:10: fatal error: SPIFFSEditor.h: No such file or directory
   10 | #include <SPIFFSEditor.h>
      |          ^~~~~~~~~~~~~~~~
compilation terminated.
exit status 1

Compilation error: SPIFFSEditor.h: No such file or directory

bmorcelli, 2025-02-21:
You need to change the version of EspAsyncWebServer to [1.3.1](https://github.com/esphome/ESPAsyncWebServer/releases/tag/v1.3.1) for it to work.

you can find this file [here](https://github.com/esphome/ESPAsyncWebServer/blob/v1.3.1/src/SPIFFSEditor.h) in this version

It may cause some other errors if you are using ESP32 v3.x framework in Arduino IDE 2.x, so you´ll need to downgrade to 2.17 or lower.

joelsernamoreno, 2025-02-21:
Hello

I am programming the firmware from scratch

In the meantime, you can use alternative firmware:

https://github.com/h-RAT/EvilCrowRF_Custom_Firmware_CC1101_FlipperZero

https://github.com/tutejshy-bit/tut-rf

If you want to use the original firmware and not waste time trying to fix the library changes, you can directly flash the .bin of version 1.3.2:

https://github.com/joelsernamoreno/EvilCrowRF-V2/blob/main/firmware/v1.3.2/EvilCrow-RFv2/EvilCrow-RFv2.ino.esp32.bin

The only difference between v1.3.2 and v1.3.3 is that v1.3.2 started to give problems with the updated libraries, I fixed this and now it is happening again

hugh-bowie, 2025-02-28:
*Awesome*.  I Had the same issue as above for about a week now. Applying corrective rollbacks now. Will advise when it works.   

**Also, id be happy to correct the minor spelling errors & update the links to the various libraries that are archived for the sake of clarity in new users. Do you take PR's from strangers?**

insideout803, 2025-03-01:
Any update on this issue? I can't get any code to flash, it always runs into some kind of error. Wethers is the SPIFFSEditor.h file missing in 1.3.3 or the AsyncElegantOTA.h file missing in 1.3.2. Not sure what to do at this point?

insideout803, 2025-03-01:
> You need to change the version of EspAsyncWebServer to [1.3.1](https://github.com/esphome/ESPAsyncWebServer/releases/tag/v1.3.1) for it to work.
>
> you can find this file [...]

This doen't work.

hugh-bowie, 2025-03-01:
@insideout803 This worked for me today, (Im on macOS Monterey 12.7.6, Intel Proc).  

- download this bin file:   [BETA_v3.3.1_EvilCrowRF_Firmware.bin](https://github.com/h-RAT/EvilCrowRF_Custom_Firmware_CC1101_FlipperZero/blob/main/Firmware/Main/EvilCrowRF/BETA_v3.3.1_EvilCrowRF_Firmware.bin)   

- download this flasher: [ESPHome-Flasher](https://github.com/esphome/esphome-flasher/releases)  ( i used the mac one  [ESPHome-Flasher-1.4.0-macOS.tar](https://github.com/esphome/esphome-flasher/releases/download/1.4.0/ESPHome-Flasher-1.4.0-macOS.tar) )  

- before flashing, i put these three folders (found in that Repo with the bin file), on the empty micro sd:  [Config+HTML+SubGhz](https://github.com/h-RAT/EvilCrowRF_Custom_Firmware_CC1101_FlipperZero/tree/main/SD)     
  
 - put that micro in the ECRF board, and plug it in to the usb, (i didnt have to push the button to go into flash mode on mac, but on windows i did)  

- Open that home flasher program .. (a caution message from OS comes up talking about its outdated. click right through that crap)  

- then flash it with the home flasher program mentioned above in bullet two. (mine worked with no messages, so i cant help past that point)

insideout803, 2025-03-02:
> [@insideout803](https://github.com/insideout803) This worked for me today, (Im on macOS Monterey 12.7.6, Intel Proc).
>
> * download this bin file: [...]

I appreciate it. I get "Unexpected error: ESP Chip Auto-Detection failed: Failed to connect to Espressif device: No serial data received." when attempting this. I guess something is wrong with my device. I may have bricked it trying all the other ways to configure it. FML

229311JK, 2025-03-20:
> Hello
>
> I am programming the firmware from scratch
>
> In the meantime, you can use alternative firmware:
>
> https://github.com/h-RAT/EvilCrowRF_Custom_Firmware_CC1101_FlipperZero [...]

Thought I'd say hello, got everything to work w/latest versions of both Arduino IDE/ESP 32 core...  
Kept it simple, w/the following steps.  1st, downloaded/extracted tagged folder.  2, followed link 2-SPIFFSEditor.h downloaded it.  3, opened INO, added SPIFFSEditor.h file.  4, removed the virtual C++ "override" comments.  5th & finally, removed line 10 <> comments & did this "SPIFFSEditor.h"...  Now, on programming in relation to this specific sketch.  The code itself is pretty clean/upfront from there if you look @it/understand programming.     When it comes 2-IDE I'd recommend enabling files in sketch option.  Likewise, going over SPIFFSEditor.CPP helps.   One thing I noticed, some of the structure originates to the ESP IDF master.  There 1 gets more insight into the C++ comments that caused the hang-up.  Here's where corporate pen-test security teams get there claims about manipulating the ESP's PHY layers in the most "pants fancy" way.  What they don't include is implementing this within the SOC environment 1 or 2 core will literally SMOKES the chip hardware!  Sorry 4-rambling but finally this too...      

Upload settings, personally I used the ESP32-PICO variant , Core debug "Error" & upload speed 921600.  It works, uploads fast.

229311JK, 2025-03-20:
From what I'm seeing of this sketch is ALOT more can be done w/it & its user friendly.  The PICO-D4 is a solid chip in the ESP32 product line.  On the other hand.  using the h-rat software puzzles me but keep this information to myself.  On h-rat, its just a bin file folks w/no published code...  Now, I'll leave all w/this riddle/question.  Why do/did all map makers leave behind tiny errors in their works??????  ; )

closed 2025-10-17 by joelsernamoreno