joelsernamoreno/EvilCrowRF-V2 #58: More options?

CommanderCrash, 2025-03-06:
I was wondering why there is no bruteforce option? Most if not all RF devices like flipper and protapack comes with these option. Is it a limitation with the hardware? I got a evil crow v2 from a firend said it was not worth the time, so i thought i can give it a try. With the advent of AI it should be kind of easy to add this in. I may take the radios out and connect them to gpio on a rpi Because that runs python. IDK the device seems kind of basic not trying to bad mouth or anything.

realdaveblanch, 2025-03-07:
You can flash

https://github.com/h-RAT/EvilCrowRF_Custom_Firmware_CC1101_FlipperZero

No SRC available for it... 
@joelsernamoreno is supposed to be working in a hidden repo rewriting the whole thing

And yeah, 0 support, gotta code it yourself

joelsernamoreno, 2025-10-17:
Hello

I’ve released a new firmware update.
I’m going to close all existing issues. If you encounter any problems with the new firmware, or if any previous issues haven’t been resolved, please open a new issue.

Thank you!

closed 2025-10-17 by joelsernamoreno