joelsernamoreno/EvilCrowRF-V2 #64: The Sketch is not working

0mk4rm4li, 2025-04-06:
After trying several ways to fix Arduino script nothing works any solution for that ? After flashing using .bin file try to visit 192.168.4.2 but nothing happened but ElegantOTA is working fine but not other

joelsernamoreno, 2025-04-06:
Hello @0mk4rm4li 

If you can access the Wi-Fi AP and you can access Elegant OTA from the web panel it means the firmware is working

All other functionality is stored on the microSD card (HTML). Have you correctly copied the folders to the microSD and inserted the microSD into the device?

If this still does not work, try using another microSD. Some microSD cards do not work. Do not use microSD cards larger than 32gb

I use this:

[image]

0mk4rm4li, 2025-04-06:
It's worked, thanks

joelsernamoreno, 2025-10-17:
Hello

I’ve released a new firmware update.
I’m going to close all existing issues. If you encounter any problems with the new firmware, or if any previous issues haven’t been resolved, please open a new issue.

Thank you!

closed 2025-10-17 by joelsernamoreno