joelsernamoreno/EvilCrowRF-V2 #68: A little clarification on SDR Firmware

SoLongSidekick, 2025-08-06:
Heya,

So stoked to see this variant of the firmware, I just have a couple questions about using it. First, when running the commands via the serial monitor what (if anything) is expected to be read out in the monitor? The first time I gave it a try I got crickets and the second time I got a bunch of seemingly nonsensical symbols. Should I expect to see what's listed in the expected output for a successful startup? Do I need to run those 4 commands every time I want to use the Crow with URH?

Regarding running the URH bridge: I tried opening a PowerShell window where the python file is stored on my PC and running the start command but I get the error `PS F:\Code\Evil Crow\SDR Firmware> python3 urh_compatible_bridge.py
Python was not found; run without arguments to install from the Microsoft Store, or disable this shortcut from Settings > Apps > Advanced app settings > App execution aliases.` so I assume I'm missing something that's going to make me feel stupid for not figuring it out. When I tried connecting to the Crow with URH I got an error telling me the connection was actively denied by the device, but I'm sure that's because the bridge isn't running.

Could you just provide some general clarifications on exactly how to do the following? 
Ensure bridge is running on port 1234 (how?)
Check serial port (/dev/cu.usbserial-140) (how? in the serial monitor? with what command?)
Verify CC1101 initialization in serial output (what's the command for this?)

I am seeing the access point so at least I know that's working correctly haha. I also see a note about removing the SD card as a troubleshooting step: should the SD slot always be vacant when using this firmware, or only if issues pop up?

Thanks in advance, I know it's a lot.

joelsernamoreno, 2025-08-08:
Hello @

I believe the SDR firmware is not functioning properly. This firmware was submitted by someone, and I accepted the PR assuming it was working correctly. However, when I tested it, I encountered the same issue — when running URH, you only see noise and strange symbols.

I have tried contacting the person who submitted the firmware, but they have not provided a solution.

If you want, you can try reaching out to them yourself in the following thread:

https://github.com/joelsernamoreno/EvilCrowRF-V2/pull/67

Sorry about this, I haven't had time to check what the problem is with this firmware

Thanks

joelsernamoreno, 2025-08-13:
Hello @SoLongSidekick 

It depends on what you want to do. If you want everything centralized in a single firmware, you can use the H-RAT firmware

If you don’t need that many features, you can find most of the components separately on my GitHub. You could also use the official firmware (though it’s less powerful) or build a custom one

Most of the time, I program a basic firmware without a web panel, tailored to whatever I need at the moment